sharebrained/portapack-hackrf #129: Streaming broken when in HackRF mode.

jboone, 2018-12-15:
@TheCranston reports: "this code has been working fine while in Portapack mode, but once i switch into hackrf mode I am unable to receive RF stream data from the unit. hackrf_info, hackrf_debug, and the other tools recognize it, but SDR apps are not receive a stream from it. reversion to hackrf released firmware and data begins to stream once more. Happy to help with any testing. thanks."

jboone, 2018-12-16:
@TheCranston I have a build that tests well here. Please give it a try and let me know if it addresses your issue: https://portapack-h1-builds.s3.amazonaws.com/sharebrained/portapack-hackrf/176/176.1/build/firmware/portapack-h1-firmware-90f64a6.tar.bz2

TheCranston, 2018-12-17:
@jboone I gave the version you linked a try and in hackrf mode I still do not get any streaming.  hackrf_sweep reports it could not transfer any data for one second and errors out.  hackrf_info reports the device.  in portapack mode it shows git version ending in 64a6.  pushing @mossmann's 2018.01.1 firmware and I'm back in business.  My TCXO is glued to the bottom of the main hackrf board with jumpers soldered so it's on and active as soon as +5v hits the board.  The si5351c does show that the external CLK is selected [0] -> 0x01.

jboone, 2018-12-18:
@TheCranston Blast. OK. To summarize the situation:

* If you load HackRF firmware into flash, everything works as expected.
* If you use the latest PortaPack firmware release or the build I provided you a few days ago, PortaPack mode works fine. HackRF mode apparently does not return samples to a USB-connected host, but all non-streaming HackRF command-line tools function as expected.

Can you remind me of the revision of your PortaPack (eight digit number, YYYYMMDD, printed near the LCD panel), and what operating system you're using for HackRF stuff? This interim firmware wasn't regression-tested with pre-2017 PortaPacks, though I'm hard-pressed to think of why that would cause an issue.

Also, can you re-program the CPLD flash bitstream with the file provided in the HackRF 2018.01.1 release, just in case it got out of sync? Though that wouldn't explain why HackRF from PortaPack mode doesn't work while the HackRF-only flash image works fine... It has to be a hardware initialization issue.

Latest firmware build is here, but there aren't any changes that would obviously impact your problem: https://s3.amazonaws.com/portapack-h1-builds/sharebrained/portapack-hackrf/178/178.1/build/firmware/portapack-h1-firmware-2c3a0c4.tar.bz2

TheCranston, 2018-12-19:
@jboone The portapack is a 20170522 reversion unit.  I'm using both Win10 and Ubuntu, primarily Ubuntu for the tools.
Your summary of the conditions are exactly correct.  The cli tools all are happy to talk with the hackrf, but anything that opens up a stream fails.

The PortaPack Setup -> Frequency Correction screen indicates "External" and "10000000"

I've re-programed the CPLD a few times, but gave it another shot based on your request.  Now it works..  something must have borked up the CPLD..  

I'll continue to test, but I think the issue is resolved..  thanks!

jboone, 2018-12-19:
Well... w00t! Please let me know how the testing goes. I'll close this in a week or so if I don't hear from you. Thanks for testing! And this is another +1 for both PortaPack and HackRF firmware doing some CPLD validation.

jboone, 2019-01-15:
Seems resolved. Closing.

closed 2019-01-15 by jboone

TheCranston, 2019-01-15:
Concur, it's goodness.  sorry for the delay in confirmation.  issue resolved.