sharebrained/portapack-hackrf #130: make a release

ZeroChaos-, 2019-01-07:
tag something and ship some binaries for your adoring fans please

jboone, 2019-01-08:
Still working on stabilizing some large changes without bricking hardware en masse. Should be this week.

jboone, 2019-01-28:
Several reports of the current commit working OK, both here and in the HAVOC fork. Still trying to address the CPLD changes, because not warning/reminding users of a CPLD bitstream change has caused utter chaos and confusion. A mechanism is forthcoming to make it abundantly clear when the CPLD is not the correct version.

jboone, 2019-01-28:
And by "HAVOC fork", I mean my fork of HAVOC. Furrtek hasn't merged my changes yet.

ZeroChaos-, 2019-02-07:
bump :-)  how we doing?

ZeroChaos-, 2019-03-11:
hi

ZeroChaos-, 2019-03-11:
you know what is better than less releases? more releases.

furrtek, 2019-05-03:
It's the exact opposite with empty messages ;p

furrtek, 2019-05-03:
@ZeroChaos- It's the exact opposite with empty messages ;p

jboone, 2019-05-07:
Release released: https://github.com/sharebrained/portapack-hackrf/releases/tag/20190405

closed 2019-05-07 by jboone