sharebrained/portapack-hackrf #132: Build failing on hackrf?

unixpunk, 2019-01-16:
So, after having trouble trying to merge things into havoc, because I'm a git novice, I instead cloned your repo and did the usual build dir cmake stuff, but make firmware is failing for me on the hackrf stuff.  Not sure what I'm missing, I have independently cloned and built the hackrf stuff without issue so I know I have all the req's.  (I also never had this issue when building havoc, seems it doesn't do this step)
`
Scanning dependencies of target hackrf
[  0%] Creating directories for 'hackrf'
[  0%] Performing download step (git clone) for 'hackrf'
Cloning into 'hackrf'...
Note: checking out 'origin/portapack_ui'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 3932c56... PortaPack: Add .SVF of current JEDEC bitstream file.
Submodule 'firmware/libopencm3' (https://github.com/mossmann/libopencm3.git) registered for path 'firmware/libopencm3'
Submodule 'hardware/gsg-kicad-lib' (https://github.com/greatscottgadgets/gsg-kicad-lib.git) registered for path 'hardware/gsg-kicad-lib'
Cloning into 'firmware/libopencm3'...
Submodule path 'firmware/libopencm3': checked out 'c9f40aae0ba25f1ffb7f2a49e050fd50714d0632'
Cloning into 'hardware/gsg-kicad-lib'...
Submodule path 'hardware/gsg-kicad-lib': checked out 'e02e7855c2f9b5653b958dab6d4e0d97eade7166'
[  0%] No patch step for 'hackrf'
[  1%] Performing update step for 'hackrf'
Current branch HEAD is up to date.
[  1%] Performing configure step for 'hackrf'
CMake Error: The source directory "/home/steve/Desktop/SDR/portapack-hackrf/build/firmware/hackrf-prefix/src/hackrf" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
firmware/CMakeFiles/hackrf.dir/build.make:105: recipe for target 'firmware/hackrf-prefix/src/hackrf-stamp/hackrf-configure' failed
make[2]: *** [firmware/hackrf-prefix/src/hackrf-stamp/hackrf-configure] Error 1
CMakeFiles/Makefile2:93: recipe for target 'firmware/CMakeFiles/hackrf.dir/all' failed
make[1]: *** [firmware/CMakeFiles/hackrf.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
`

unixpunk, 2019-01-16:
To point out, the message is correct, hackrf does not contain CMakeLists.txt, only the firmware, tools folders do.

unixpunk, 2019-01-20:
If you want to fix on this with your other CPLD work on Monday, I found a fix after running cmake.  I edited firmware/CMakeFiles/hackrf.dir/build.make in my build directory and changed line 106 from 
`cd /my/build/path/portapack-havoc/build/firmware/hackrf-prefix/src/hackrf-build && /usr/bin/cmake -DUSER_INTERFACE=PORTAPACK SOURCE_SUBDIR firmware/hackrf_usb "-GUnix Makefiles" /my/build/path/portapack-havoc/build/firmware/hackrf-prefix/src/hackrf`

To instead be:
`cd /my/build/path/portapack-havoc/build/firmware/hackrf-prefix/src/hackrf-build && /usr/bin/cmake -DUSER_INTERFACE=PORTAPACK SOURCE_SUBDIR firmware/hackrf_usb "-GUnix Makefiles" /my/build/path/portapack-havoc/build/firmware/hackrf-prefix/src/hackrf/firmware/hackrf_usb`
Adding in the missing ending `/firmware/hackrf_usb`.  I'm just not sure how to do this in the cmake files BEFORE running cmake so I won't attempt a pull request.  Based on this I'm guessing you'll know where to properly make the change.  Thanks for all the work on this!  Let me know if I can (try to) help further!

unixpunk, 2019-01-23:
FYI, still no dice here, actually worse now with latest pull from jboone/portapack-havoc, I can't evenworkaround it now.  Maybe I'm missing something?  git clone;mkdir build;cd build;cmake ..;make firmware -> fail within hackrf again, this time other errors even after I fix the above as described.  I have the same issue with portapack-hackrf after latest pull as well.

[ 26%] No patch step for 'hackrf'
[ 26%] Performing update step for 'hackrf'
Current branch master is up to date.
[ 27%] Performing configure step for 'hackrf'
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /usr/arm-none-eabi
CMake Error at /home/user/Desktop/SDR/portapack-havoc/build/firmware/hackrf-prefix/src/hackrf/firmware/hackrf-common.cmake:30 (include):
  include could not find load file:

    /dfu-util.cmake
Call Stack (most recent call first):
  CMakeLists.txt:26 (include)

CMake Error at /home/user/Desktop/SDR/portapack-havoc/build/firmware/hackrf-prefix/src/hackrf/firmware/hackrf-common.cmake:105 (include_directories):
  include_directories given empty-string as include directory.
Call Stack (most recent call first):
  CMakeLists.txt:26 (include)

CMake Error: File /m0_bin.s.cmake does not exist.
CMake Error at /home/user/Desktop/SDR/portapack-havoc/build/firmware/hackrf-prefix/src/hackrf/firmware/hackrf-common.cmake:149 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  CMakeLists.txt:57 (DeclareTargets)

-- Configuring incomplete, errors occurred!
See also "/home/user/Desktop/SDR/portapack-havoc/build/firmware/hackrf-prefix/src/hackrf-build/CMakeFiles/CMakeOutput.log".
See also "/home/user/Desktop/SDR/portapack-havoc/build/firmware/hackrf-prefix/src/hackrf-build/CMakeFiles/CMakeError.log".
make[3]: *** [firmware/hackrf-prefix/src/hackrf-stamp/hackrf-configure] Error 1
firmware/CMakeFiles/hackrf.dir/build.make:105: recipe for target 'firmware/hackrf-prefix/src/hackrf-stamp/hackrf-configure' failed
make[2]: *** [firmware/CMakeFiles/hackrf.dir/all] Error 2
CMakeFiles/Makefile2:93: recipe for target 'firmware/CMakeFiles/hackrf.dir/all' failed
make[1]: *** [firmware/CMakeFiles/firmware.dir/rule] Error 2
CMakeFiles/Makefile2:138: recipe for target 'firmware/CMakeFiles/firmware.dir/rule' failed
make: *** [firmware] Error 2
Makefile:131: recipe for target 'firmware' failed

I also get it trying to do what I think your cmake is doing:
user@pc:~/Desktop/SDR/portapack-hackrf/build/firmware/hackrf-prefix/src/hackrf-build$ cmake ../hackrf/firmware/hackrf_usb

jboone, 2019-01-23:
OK, I couldn't let this go. So I believe I fixed it both in this repo, and in my fork of portapack-havoc. I even tested it first! Give it another try and let me know how it goes.

jboone, 2019-01-24:
I'm stumped -- it's built clean for me and a couple other people since yesterday.

Can you capture a complete log of commands and output, starting from nothing, and drop it on PasteBin? Having a complete transcript of the process should help me quickly reproduce the issue and identify the problem. Your operating system and CMake version might also come in handy.

jboone, 2019-01-24:
So, one thing I've observed: The ExternalProject_Add CMake command doesn't support SOURCE_SUBDIR until version 3.7. That's what allows me to dig into the HackRF firmware subdirectory and build from there. I'll look for an alternative, since it'd be a bummer to require 3.7 when Ubuntu 16.04 has 3.5. Ubuntu 16.04 isn't *that* old...

unixpunk, 2019-01-24:
Ah, interesting, nice find!  Ya, 16.04 LTS still supported.  :)

unixpunk, 2019-01-24:
I miss the good ole days of ./configure && make...cmake seems so overly complicated sometimes.  :)

jboone, 2019-01-24:
I'm looking for a fix. Just using SOURCE_DIR doesn't do it. The best solution may be to introduce a top-level CMakeLists.txt for HackRF. I'm not sure why it doesn't have one.

> I miss the good ole days of ./configure && make...cmake seems so overly complicated sometimes. :)

Agreed. It helped with some host operating system complications on the HackRF project -- like not needing to know how to build on Windows. But it's complicated and weird and inconsistent in so many ways. It may be more trouble than it's worth in this case, but I'm not about to upend things again right now. :-)

unixpunk, 2019-01-24:
The only thing you're building is hackrf/firmware/hackrf_usb so if you can get your code to go straight there instead, that should work, aside from maybe my other comment about the variable and /dfu-util.cmake not being found.  Before you tried to fix this, i was able to workaround it by editing the file as noted above, but after your attempt, I could no longer even do that.  Not sure if those details help or not.  Referring to my comment on the 20th.  Whatever was changed after that broke my workaround too.  Anyway, no rush, done for the night.  Thanks again!

jboone, 2019-01-24:
I made a change 13 hours ago that does not use the SOURCE_SUBDIR variable introduced in CMake 3.7. I've tested the change locally and on Travis-CI, and it appears to work. Please pull a fresh copy of the master branch of this repo and let me know if it works for you.

unixpunk, 2019-01-25:
You're the man!  That did it!  Tested both portapack-hackrf and your fork of portapack-havoc.  Thanks again for knocking this out!

closed 2019-01-25 by unixpunk