sharebrained/portapack-hackrf #134: I update the firmware, ,but the screen is always blank.

freeworld1981, 2019-02-27:
I updated the firmware and CPLD, all successfully ,but the screen is always blank, dark no light. 

the led only green/yellow/red light.

where is the problem?

jboone, 2019-03-21:
@freeworld1981 It's hard to say. Which LEDs are lit? Just the 3V3, 1V8, and RF power LEDs? Or are the USB (green), RX (yellow), or TX (red) LEDs also lit?

Which version of PortaPack firmware are you using? Is it a released version, or are you compiling it yourself?

Which PortaPack hardware revision are you using? There's a date code on the front of the PortaPack, near the LCD panel.

pipi111, 2019-03-29:
same here i have githus 5ec8164e
latest portapack hardare version
latest firmware
downloaded from original source

kevtheskin, 2019-04-26:
Hello Jared, Im have the same problem. I tried the havoc software and i think my unit is a brick now?. Cheers Kev

jboone, 2019-05-07:
@kevtheskin, @pipi111, @freeworld1981: Please try the latest released PortaPack firmware and report your results: https://github.com/sharebrained/portapack-hackrf/releases/tag/20190405

jboone, 2019-11-16:
Are your problems resolved by using the release firmware?

jboone, 2020-01-24:
No action on this thread. Closing.

closed 2020-01-24 by jboone

Mars0117, 2020-02-26:
Hi.
I faced that problem. Original firmware not working https://github.com/sharebrained/portapack-hackrf/releases/tag/20190405.
Before the problem the original firmware and havoc worked.

millercontracting2017, 2021-04-06:
Same problem here also any fix yet?