sharebrained/portapack-hackrf #138: No HackRF boards found

xvostikk, 2019-05-09:


stephantual, 2019-06-01:
Here's a bit of help - as i struggled with the same issue.
I imagine  you find yourself with a box in a 'dead' state, with a black screen, so hackrf_spi can't find the board but putting the portapack/hackrf in dfu mode allows you to list it.

So, to solve this, this is what i did:
- dfu upload the standard h1, this should be a good basis to get started
- spi flash the standard portapack . this gives you back access to the hackrf_spi tool
- now, put the portapack in standard hackrf mode (important step!)
- finally, unplug, put in dfu, plug back and DO NOT use the spi tool to flash to havoc, instead, go to your  havoc build folder and 'make program'. 
- the above step might take a few tries to get right - because if it fails at any point, it bounces your device out of dfu and back to black screen. So try the above steps until make program executes without error-
- once you have acheived an error less 'make program', it will detect your dfued hack rf, and you will see the following (if all goes well):

`[100%] Built target application.elf
[100%] Built target application
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download        [=========================] 100%        35188 bytes
Download done.
dfu-util: unable to read DFU status after completion
File size 781884 bytes.
Checking target device compatibility
Erasing SPI flash.
Writing 781884 bytes at 0x000000.
[100%] Built target program
`

In other words, you are probably missing two steps; a) you need to put the portapack in hackrf mode b) don't try to spi flash havoc, instead, make program while in dfu 

i hope this helps!

xvostikk, 2019-07-24:
Thanks for the reply. I did exactly what you have written. So after i put the device in dfu Mode and make command ```make program``` i had the output:
```
user@user:~/Git/portapack-havoc/build$ make program
[ 25%] Built target baseband_shared
[ 25%] Built target baseband_nfm_audio.elf
[ 25%] Performing update step for 'hackrf'
Aktueller Branch master ist auf dem neuesten Stand.
[ 26%] Performing configure step for 'hackrf'
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /usr/arm-none-eabi
-- Configuring done
-- Generating done
-- Build files have been written to: /home/administrator/Git/portapack-havoc/build/firmware/hackrf-prefix/src/hackrf-build
[ 26%] Performing build step for 'hackrf'
[ 15%] Built target libopencm3_hackrf_usb
[ 19%] Built target hackrf_usb_m0.elf
[ 19%] Built target hackrf_usb_m0.bin
[100%] Built target hackrf_usb_dfu_objects
[100%] Built target hackrf_usb_dfu.elf
[100%] Built target hackrf_usb_dfu.bin
dfu-suffix (dfu-util) 0.9

Copyright 2011-2012 Stefan Schmidt, 2013-2014 Tormod Volden
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Suffix successfully added to file
[100%] Built target hackrf_usb.dfu
[ 26%] No install step for 'hackrf'
[ 26%] Completed 'hackrf'
[ 27%] Built target hackrf
[ 27%] Built target baseband_wfm_audio.elf
[ 28%] Built target baseband_wideband_spectrum.elf
[ 28%] Built target baseband_sstvtx.elf
[ 29%] Built target baseband_test.elf
[ 29%] Built target baseband_siggen.elf
[ 30%] Built target baseband_pocsag.elf
[ 31%] Built target baseband_no_operation.elf
[ 32%] Built target baseband_tpms.elf
[ 32%] Built target baseband_afsktx.elf
[ 32%] Built target baseband_rds.elf
[ 33%] Built target baseband_afskrx.elf
[ 34%] Built target baseband_fsktx.elf
[ 34%] Built target baseband_tones.elf
[ 34%] Built target baseband_adsbtx.elf
[ 35%] Built target baseband_adsbrx.elf
[ 35%] Built target baseband_ook.elf
[ 35%] Built target baseband_ais.elf
[ 36%] Built target baseband_am_audio.elf
[ 36%] Built target baseband_audio_tx.elf
[ 37%] Built target baseband_capture.elf
[ 37%] Built target baseband_jammer.elf
[ 37%] Built target baseband_replay.elf
[ 37%] Built target baseband_ert.elf
[ 38%] Built target baseband_sonde.elf
[ 38%] Built target baseband_acars.elf
[ 39%] Built target baseband_mic_tx.elf
[ 46%] Built target baseband
[100%] Built target application.elf
[100%] Built target application
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[=========================] 100%        35188 bytes
Download done.
dfu-util: unable to read DFU status after completion
File size 767068 bytes.
hackrf_open() failed: HackRF not found (-5)
firmware/CMakeFiles/program.dir/build.make:57: recipe for target 'firmware/CMakeFiles/program' failed
make[3]: *** [firmware/CMakeFiles/program] Error 1
CMakeFiles/Makefile2:127: recipe for target 'firmware/CMakeFiles/program.dir/all' failed
make[2]: *** [firmware/CMakeFiles/program.dir/all] Error 2
CMakeFiles/Makefile2:134: recipe for target 'firmware/CMakeFiles/program.dir/rule' failed
make[1]: *** [firmware/CMakeFiles/program.dir/rule] Error 2
Makefile:131: recipe for target 'program' failed
make: *** [program] Error 2
```
For me seems like the program change then hackrf Mode and can not find the device.
After some tryings i did as it stays here: https://github.com/furrtek/portapack-havoc/wiki/Flashing:
"Alternatively, the HackRF may be flashed using the HackRF mode from PortaPack's menu:
```hackrf_spiflash -w portapack-h1-havoc.bin```". Of course now in hackrf Mode.

I am really irritated what is the right Method and why it doesn't work in other way. But it seems working.

Thank you for the answer!

jboone, 2019-11-16:
I'm sorry for the delay.

I am wondering if the delay in the firmware/CMakeLists.txt:program target is sufficient on your machine. Perhaps it takes your computer more than three seconds to enumerate the HackRF once it leaves DFU mode and enters HackRF mode. Try changing the "sleep 3s" to "sleep 15s" and see if that has an effect?

jboone, 2020-02-05:
No response. Closing.

closed 2020-02-05 by jboone