sharebrained/portapack-hackrf #142: Firmware build error

uglurass, 2019-08-02:
anyone make sense of this ..
[ 35%] Performing build step for 'hackrf'
Scanning dependencies of target libopencm3_hackrf_usb
[  0%] Creating directories for 'libopencm3_hackrf_usb'
[  3%] No download step for 'libopencm3_hackrf_usb'
[  7%] No patch step for 'libopencm3_hackrf_usb'
[  7%] No update step for 'libopencm3_hackrf_usb'
[  7%] No configure step for 'libopencm3_hackrf_usb'
[ 11%] Performing build step for 'libopencm3_hackrf_usb'
  GENHDR  include/libopencm3/efm32/efm32lg/irq.yaml
/usr/bin/env: ‘python’: No such file or directory
Makefile:58: recipe for target 'include/libopencm3/efm32/efm32lg/irq.yaml.genhdr' failed
make[8]: *** [include/libopencm3/efm32/efm32lg/irq.yaml.genhdr] Error 127
hackrf_usb/CMakeFiles/libopencm3_hackrf_usb.dir/build.make:110: recipe for target 'hackrf_usb/libopencm3_hackrf_usb-prefix/src/libopencm3_hackrf_usb-stamp/libopencm3_hackrf_usb-build' failed
make[7]: *** [hackrf_usb/libopencm3_hackrf_usb-prefix/src/libopencm3_hackrf_usb-stamp/libopencm3_hackrf_usb-build] Error 2
CMakeFiles/Makefile2:955: recipe for target 'hackrf_usb/CMakeFiles/libopencm3_hackrf_usb.dir/all' failed
make[6]: *** [hackrf_usb/CMakeFiles/libopencm3_hackrf_usb.dir/all] Error 2
CMakeFiles/Makefile2:706: recipe for target 'hackrf_usb/CMakeFiles/hackrf_usb.dfu.dir/rule' failed
make[5]: *** [hackrf_usb/CMakeFiles/hackrf_usb.dfu.dir/rule] Error 2
Makefile:313: recipe for target 'hackrf_usb.dfu' failed
make[4]: *** [hackrf_usb.dfu] Error 2
firmware/CMakeFiles/hackrf.dir/build.make:110: recipe for target 'firmware/hackrf-prefix/src/hackrf-stamp/hackrf-build' failed
make[3]: *** [firmware/hackrf-prefix/src/hackrf-stamp/hackrf-build] Error 2
CMakeFiles/Makefile2:160: recipe for target 'firmware/CMakeFiles/hackrf.dir/all' failed
make[2]: *** [firmware/CMakeFiles/hackrf.dir/all] Error 2
CMakeFiles/Makefile2:210: recipe for target 'firmware/CMakeFiles/firmware.dir/rule' failed
make[1]: *** [firmware/CMakeFiles/firmware.dir/rule] Error 2
Makefile:157: recipe for target 'firmware' failed
make: *** [firmware] Error 2

jboone, 2019-08-04:
These errors look to me like you might not have Python 2 installed. What operating system and distribution are you using? Here are the requirements for the libopencm3 project, which appears to be the part of your build that is breaking: https://github.com/libopencm3/libopencm3 This closed issue might also be relevant: https://github.com/libopencm3/libopencm3/issues/1075

Please let us know if this helps.

uglurass, 2019-08-05:
i am trying to do this from havoc instructions ubuntu 16.04 xenial. is there any better platform?

uglurass, 2019-08-07:
it was a missing pyyaml 
here is a working procedure 
sudo apt-get install git 
sudo apt-get install cmake 
sudo apt-get install dfu-util

$ wget https://bootstrap.pypa.io/get-pip.py
$ sudo python2.7 get-pip.py
which pip2.7
sudo pip install pyyaml
sudo add-apt-repository ppa:team-gcc-arm-embedded/ppa
sudo apt-get update
sudo apt-get install gcc-arm-embedded
git clone https://github.com/furrtek/portapack-havoc.git
cd portapack-havoc
mkdir build
cd build
cmake ..
make firmware

jboone, 2019-11-16:
Wiki updated to mention PyYAML dependency. Thanks!

closed 2019-11-16 by jboone