sharebrained/portapack-hackrf #145: reflash firmware using windoes?

kiwiandy, 2019-11-28:
I have uploaded hackrf_one_usb.dfu to my device using DF-UTIL command line on my Windows PC.

How do I get get the Portapack code flashed into the SPI flash?

Macintoshsh, 2019-11-29:
https://github.com/sharebrained/portapack-hackrf/wiki/Updating%20Firmware%20Windows

Sent from  iPhone A_Macintosh
nr +79268246716

kiwiandy, 2019-11-29:
> https://github.com/sharebrained/portapack-hackrf/wiki/Updating%20Firmware%20Windows Sent from iPhone A_Macintosh nr +79268246716
> […](#)

Thanks but as my screen is blank on the Portapack, I cannot set the portapack to HACKRF mode so that I can use that tool!

Macintoshsh, 2019-11-29:
So it’s tool don’t using screen portapack. It’s flashing from computer. 
I’m flash from Ubuntu and dfu utils. 
I’m connect and flashing HACKRf  and  portapack. 
After flashing all start. 

Sent from  iPhone A_Macintosh
nr +79268246716

> 29 нояб. 2019 г., в 21:16, kiwiandy <notifications@github.com> написал(а):
> 
> Thanks but as my screen is blank on the Portapack, I cannot set the portapack to HACKRF mode so that I can use that tool

jboone, 2020-01-24:
Good answer. A demo of how to do Windows flashing is here: https://www.youtube.com/watch?v=5G5blPPIIBo

Since there was no reply, I'm going to call this one closed.

closed 2020-01-24 by jboone