sharebrained/portapack-hackrf #148: QUESTION - SWR meter option possible?

DinBan, 2020-01-21:
Hello, is there any way to add to Portapack firmware SWR/VNA menu/option in future?
I found that HackRF/SDR can be user as SWR meter with cheap noise source -
https://www.rtl-sdr.com/measuring-the-swr-of-fpv-antennas-with-an-rtl-sdr/
https://www.rtl-sdr.com/rtl-sdr-tutorial-measuring-filter-characteristics-and-antenna-vswr-with-an-rtl-sdr-and-noise-source/
Also found on GitHub some hackrf_antenna_swr_utility -
https://github.com/coddingtonbear/hackrf-antenna-swr-utility/tree/master/hackrf_antenna_swr_utility
I think it would be nice and useful option if it possible.
Thank you!

DinBan, 2020-02-05:
Any answer? Hello?

jboone, 2020-02-05:
I agree it would be useful. I only think an SWR meter is feasible. A VNA would require two HackRFs and some extensive hardware modifications.

For an SWR meter using a noise source, it sounds like what is needed is a baseline difference function on the spectrum display. A sweeping spectrum analyzer like what is done with hackrf_sweep would be useful to extend the frequency range of the analyzer beyond the 20MHz bandwidth of the baseband ADCs.