sharebrained/portapack-hackrf #149: HackRF mode: gains displayed incorrectly during T/R switching

argilo, 2020-01-27:
I'm working on adding transmit/receive switching capability into gr-osmosdr (https://github.com/osmocom/gr-osmosdr/compare/gr3.8...argilo:hackrf-allow-tr-switching-38), and noticed that the Portapack does not correctly update the displayed gains when switching between transmit & receive mode.

When returning from transmit to receive mode, the BB gain display remains blank until the next time it is adjusted. When switching in either direction, the IF gain display does not update unless the TX IF gain or RX IF gain were adjusted prior to the switch.

One possible solution would be to update all gain displays whenever switching from transmit to receive or vice versa.

jboone, 2020-02-04:
Argh. Thanks for the heads-up. I will take a look at it, hopefully before GSG ships their firmware release, as the affected code is actually in the HackRF repo.

argilo, 2020-02-05:
In case it helps with testing, the T/R switching code is now merged into the gr3.8 branch of gr-osmosdr:

http://git.osmocom.org/gr-osmosdr/commit/?h=gr3.8&id=42ed7c4d99809c619fb386d89fabd82509901656

jboone, 2020-02-08:
Well, I finally got a working GR 3.8 install. *whew* I'll have a look tomorrow.