sharebrained/portapack-hackrf #150: Touch Screen

Jean-france, 2020-02-03:
Hi,
I have HAVOC firmware on my HK. The touch screen doesn't work. Impossible to calibrate it, nothing do even i touch the screen.
Do you have a idea ?
Many thanks in advance.
Best regards,
Jean

jboone, 2020-02-04:
If you go into "Receivers", "Audio", highlight the frequeny, and press the select button, you will get a number entry screen. If you touch the screen, does the PortaPack respond to your touches?

Jean-france, 2020-02-05:
Thanks for your reply.

No, nothing, no response.
Same thing if i go to setting the touch screen, no response if i touch the cross

From: Jared Boone 
Sent: Tuesday, February 04, 2020 11:50 PM
To: sharebrained/portapack-hackrf 
Cc: Jean-france ; Author 
Subject: Re: [sharebrained/portapack-hackrf] Touch Screen (#150)

If you go into "Receivers", "Audio", highlight the frequeny, and press the select button, you will get a number entry screen. If you touch the screen, does the PortaPack respond to your touches?

jboone, 2020-02-05:
Can you tell me which version of HAVOC you are using? You can find the version in the "About" item in the main menu.

Does the problem exist if you use my "official" firmware? Here's a link to the latest release: https://github.com/sharebrained/portapack-hackrf/releases/tag/20190405

Where did you buy this PortaPack? Which version of hardware is it? The hardware version number is written on the circuit board, along the top-left edge of the LCD screen.

Jean-france, 2020-02-06:
Version of HAVOC : Git hash dc5d6fe
I don’t know i didn’t try with “official” firmware.
I bought my PortaPack on Ebay
version number : 20181029 

Thanks for your help.

From: Jared Boone 
Sent: Wednesday, February 05, 2020 7:14 PM
To: sharebrained/portapack-hackrf 
Cc: Jean-france ; Author 
Subject: Re: [sharebrained/portapack-hackrf] Touch Screen (#150)

Can you tell me which version of HAVOC you are using? You can find the version in the "About" item in the main menu.

Does the problem exist if you use my "official" firmware? Here's a link to the latest release: https://github.com/sharebrained/portapack-hackrf/releases/tag/20190405

Where did you buy this PortaPack? Which version of hardware is it? The hardware version number is written on the circuit board, along the top-left edge of the LCD screen.

jboone, 2020-02-06:
I would recommend trying a newer version of HAVOC or trying with the official firmware just to ensure there's no issue with that particular version of firmware.

I can confirm for the PortaPacks that I make that the touch screen works for the latest HAVOC and official firmware. So if those do not work for you, you probably have a hardware issue and should contact the eBay seller for a repair or replacement.

Jean-france, 2020-02-06:
OK Jared, thanks for your reply.
Can you send me the latest HAVOC firmware, please ?

From: Jared Boone 
Sent: Thursday, February 06, 2020 6:42 PM
To: sharebrained/portapack-hackrf 
Cc: Jean-france ; Author 
Subject: Re: [sharebrained/portapack-hackrf] Touch Screen (#150)

I would recommend trying a newer version of HAVOC or trying with the official firmware just to ensure there's no issue with that particular version of firmware.

I can confirm for the PortaPacks that I make that the touch screen works for the latest HAVOC and official firmware. So if those do not work for you, you probably have a hardware issue and should contact the eBay seller for a repair or replacement.

jboone, 2020-02-06:
I have continuous integration builds of the latest HAVOC code here:

https://jboone.github.io/portapack-havoc-nightly/

Jean-france, 2020-02-06:
Many thanks !
I will instal the 26/09/2019 firmware version

From: Jared Boone 
Sent: Thursday, February 06, 2020 7:21 PM
To: sharebrained/portapack-hackrf 
Cc: Jean-france ; Author 
Subject: Re: [sharebrained/portapack-hackrf] Touch Screen (#150)

I have continuous integration builds of the latest HAVOC code here:

https://jboone.github.io/portapack-havoc-nightly/

Jean-france, 2020-02-06:
Could you please send me the  link for download the latest version of ExtIO_harcrf, please.
Sorry to disturbe you.

From: Jared Boone 
Sent: Thursday, February 06, 2020 7:21 PM
To: sharebrained/portapack-hackrf 
Cc: Jean-france ; Author 
Subject: Re: [sharebrained/portapack-hackrf] Touch Screen (#150)

I have continuous integration builds of the latest HAVOC code here:

https://jboone.github.io/portapack-havoc-nightly/

jboone, 2020-02-06:
Do you mean this? https://github.com/jocover/ExtIO_HackRF

I don't know what that is, but hopefully the link helps.

Jean-france, 2020-02-11:
> https://jboone.github.io/portapack-havoc-nightly/
>
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/sharebrained/portapack-hackrf/issues/150?email_source=notifications\u0026email_token=AONV3ZKI54F3YAWXHFRCEZDRBRIK3A5CNFSM4KPIRXZKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOELAIDJY#issuecomment-583041447",
> "url": "https://github.com/sharebrained/portapack-hackrf/issues/150?email_source=notifications\u0026email_token=AONV3ZKI54F3YAWXHFRCEZDRBRIK3A5CNFSM4KPIRXZKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOELAIDJY#issuecomment-583041447",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> }
> ]

Hi,The new firmware 79baef7 doesn't work !If you choice "capture" the device is blocked! You have to disconnect it .What's wrong?Thanks in advance

Jean-france, 2020-02-11:
So sorry, the new firmware work !It's my fault... pardon.

https://jboone.github.io/portapack-havoc-nightly/

[
{

jboone, 2020-03-20:
No activity, so I will assume the issue is resolved. Closing...

closed 2020-03-20 by jboone