sharebrained/portapack-hackrf #153: Bq24075 bat charge on h1

antonlalala, 2020-03-20:
Hello friends, are there people who soldered bq24075? On h1 
[image]
[image] 
How will it even turn on? by pressing which button? I did not understand.

jboone, 2020-03-20:
Hi. Thank you for opening the issue here.

The news is not good. The battery feature was an experiment, and was never turned into a finished feature. So there is no way to turn the power on or off from the firmware.

antonlalala, 2020-03-20:
☹️😢

antonlalala, 2020-03-20:
GPS module for what purpose?

jboone, 2020-03-20:
Also an experiment. There is no firmware support for it, and I have not tested it at all.

antonlalala, 2020-03-20:
It would be nice to do GPS. Ais transponder, aprs ..

jboone, 2020-03-20:
True. There's probably also value to recording current location when logging data, and having precise time available.

antonlalala, 2020-03-20:
that would be great, as well as navigation

antonlalala, 2020-03-20:
going back to bq. If I solder it according to the scheme, then the device will be constantly on while the battery is charged? or will I ruin the device?

vasja2000, 2020-04-03:
hi! Colleagues, help me find the scheme exactly Portapack_h2. We need it.

jboone, 2020-04-03:
@vasja2000 You need to contact the people who make the H2. I have no information on the H2. It is a completely different design, as far as I can tell.

jboone, 2020-04-03:
@antonlalala Yes, the device will be on constantly, until the battery is discharged, or you plug in power to USB.

eried, 2020-04-06:
BTW the on off works perfectly on the H2, such a shame that the pcb is not available anywhere... (I wish there was a way to have battery level on the UI)

zhaohuansi, 2020-04-26:
The Chinese foundry is already produced. Give you a photo tomorrow.

cql1983, 2020-05-08:
> 
> 
> BTW the on off works perfectly on the H2, such a shame that the pcb is not available anywhere... (I wish there was a way to have battery level on the UI)

H2 doesn't exist. It's just a modified version based on H1 that a hobbyist added logic chips for switches.

eried, 2020-05-08:
> H2 doesn't exist.

On that ethereal non existant H2, on off works perfectly

cql1983, 2020-05-08:
Yes, you can visit :  https://market.m.taobao.com/app/idleFish-F2e/widle-taobao-rax/page-detail?wh_weex=true&wx_navbar_transparent=true&id=596112633971&ut_sk=1.WRF5Cr3ABR8DAMOzG71j1M%2Bu_12431167_1588928139854.Copy.detail.596112633971.139574515&forceFlush=1 
H2 is Chinese fans modify.

antonlalala, 2020-05-13:
> Привет. Спасибо за открытие вопроса здесь.
> 
> Новости не хорошие. Функция батареи была экспериментом и никогда не превращалась в готовую функцию. Так что нет возможности включить или выключить питание из прошивки.

can i put any dcdc converter between audio svdd and vbus? and switch to lipo?

zjywlive, 2020-08-03:
Hello everyone, I'm sorry this is a message translated by computer,I am familiar with this designer, and I am also involved in the design of some peripheral parts.

antonlalala, 2020-08-20:
No!