sharebrained/portapack-hackrf #155: Headphone Failure

SannhetJaeger, 2020-04-15:
Posted this initially in furrtek/portapack-havoc, but unsure if that is right place since the issue manifests on base portapack firmwares as well.

Think I am dealing with a defective portapack, but thought I would see if anyone has any ideas on steps to take to remedy or be sure its a hardware failure. Headphones produce no output, regardless of volume level. I've tried with several different firmwares (20170808, Havoc 1352e16 and a locally built Havoc from few days ago). Right now have the 20190405 installed. This is a 20181029 Blue Portapack board on a 10 Feb 2014 HackRF One. Debug of AK4951 shows all zeros(0).

Only reaction I can get is on microphone transmit - pressing the apple headset button makes the mic tx level bar come on if it wasn't there already (e.g. check / uncheck voice activated will make bar disappear). Unfortunately, that's only response - no actual mic sounds are picked up. Thought it relevant to share that oddity as it shows "something" is getting through.

The headsets are confirmed working on other devices. Used several other brands of TRRS (left, right, ground, mic) headsets. Also tried TRS (left, right, ground) to no effect. Confirmed continuity from each jack to jack mount pads to next circuit step (e.g. left-C22, right-C23, mic-C25, ground).

Thanks in advance!

jboone, 2020-04-16:
If this is not a hardware issue (PortaPack, HackRF, or headphones), the only reason I can think of would be the receiver mode. The FM receiver mode squelches when there's no or insufficient modulation present. But I imagine you've tried several different modes already.

Did this combination of hardware work at some point, or has it never worked since you received the hardware? Where did you buy the PortaPack and the HackRF?

SannhetJaeger, 2020-04-19:
Waterfall showing a solid WFM signal - cycled through other modes to no effect.  Recently received from what I understand to be reputable source on Ali.  Seller offered to exchange, but I hate to wait weeks for the process only to find it still doesn't work for some other reason (e.g. problem on HackRF board - bought bundle from seller).  Everything but headphone worked out-of-box.  

Subsequently examined board under microscope and quality looks good / clean.  Test all ground and power rails / points.  Since I don't have a working board to compare in-circuit component values, I have exhausted testing unpowered.  Have to get some ribbons to test the audio chip / circuit - with slow shipping lately, will take time to get those jumpers - I don't have enough in bin.

The upside with this is getting a deep dive into the hardware and low level side.  Though, it been drinking from the fire hose these days!

Povezi, 2020-04-19:
I it the same issue like. mine: https://github.com/sharebrained/portapack-hackrf/issues/140

I suppose it is hardware problem, because it stopped working. Any idea how to fix it?

SannhetJaeger, 2020-04-20:
@Povezi I have no sound at all from headphones.  Occasionally there will be clicking in pace with one of the clocks, which unplugging and replugging headphones will eliminate.  Typically clicks will occur after repeated plugging + unplugging.  Volume levels have no impact on headphone output in my case.

Povezi, 2020-04-20:
Yes, I had sound, but in the middle of FM listening it died. Now no sound at all. Looking for a hint how to repair hardware...

serjjjz, 2020-04-21:
There was a similar problem. The microphone was working intermittently. The reason was a bad factory weeding. Fixed with a hair dryer.
[image]
[image]

SannhetJaeger, 2020-04-24:
Finally figured out the problem.  They surface mounted the AK4951 chip rotated 90 degrees counter clockwise.  Was hiding in plain sight the whole time.  Lesson learned - check pin positions somewhere in top 3 steps; before probing I2C, I2S and reflowing board several times...  Fortunately, the rotation does not appear to have done any damage to the chip or other circuits.

closed 2020-04-24 by SannhetJaeger