sharebrained/portapack-hackrf #157: Devices detected: 0 DFU, 0 HackRF

andytang123456789, 2020-04-30:
Devices detected: 0 DFU, 0 HackRF
ERROR: HackRF not found.
Please connect a HackRF One to your computer's USB port.
If using a PortaPack, please put it in HackRF mode after connecting it.
but i  am sure i have connect the hackrf one and i can  receive the radio through #SDR, but I cannot flash the firmware because the firmware cannot recognize hackrf one, even if I enter hackrf mode with portapack

Ghettopeter, 2020-05-02:
I have the same problem. I also tried putting the HackRF device in DFU mode but that did not help at all.  SDR# works fine using the same cable and the same USB port.  I am using Windows10.

philsalesses, 2020-06-06:
I am on Mac OSX 10.15.4 Catalina and have the same issue.

angelfr30, 2020-06-16:
Same problem with Windows 10.

saltchicken, 2020-09-25:
I am having the same issue on Windows 10. I can use the HackRF in SDR# when the portapack is in HackRF mode, but I am unable to flash the device. Same issue even with DFU mode.

JCarter0525, 2020-11-10:
Has anyone found a resolution to this issue yet?

rgray107918, 2020-11-28:
Same issue here.  I tried running as administrator, no dice.  I even removed the portapack board, and tried entering DFU mode in the HackRF alone, and get same issue.

I did hold down the Reset + DFU button for closer to 15 seconds, and when running the DFU file, it gave me this:

*** Run HackRF firmware in RAM via LPC DFU ***

This is used to "unbrick" your HackRF, if you are no longer able to use
HackRF tools to flash or operate your HackRF.

Connect your HackRF One to a USB port on your computer.

Hold down both the DFU and RESET buttons on the HackRF.
Then release the RESET button (closest to the edge).
Then release the DFU button.

Press any key to continue . . .

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

DFU suffix CRC does not match
A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download        [=========================] 100%        22056 bytes
Download done.
unable to read DFU status after completion
can't detach
Resetting USB to switch back to runtime mode

Press any key to continue . . .

My device also works fine in SDR#.  Interestingly, it will not run on SDRAngel.  However, I don't know if that is a device issue or an SDRAngel issue.

MichaelSMVogel, 2020-12-03:
It's a driver issue - I resolved it by finding the hackrf device in the device manager and manually updating the driver. You can point it to the directory where you have the firmware downloaded. There is a driver folder. Windows tells you it already has the best driver but ignore that and choose from list - have disk - there are likely at least two versions in the list. It worked after I switched to the 2015 version.

JCarter0525, 2020-12-03:
I'll check it out and let you know. Thanks for letting me know the workaround!

JCarter0525, 2020-12-13:
I'm beginning to wonder if I have a defective device. I'm to the point that I'm willing to send it to someone who can complete this and pay them for their time...

MichaelSMVogel, 2020-12-13:
Did you try the driver change? I have experienced this twice now on 3 computers. Windows found a newer driver and installed it on it's own and it works as far as the hackrf is concerned bug the firmware utility can not see the hackrf.

RAMOhio, 2020-12-25:
I solved the issue by removing the PortaPack from the HackRF, flashing the firmware with the PortaPack removed, and then once that completed everything was fine.

Additional notes:
- Windows detected the HackRF only when the PortaPack was not plugged in.
- I had the battery connected to the PortaPack (not sure if this makes a difference, probably not, but I'm adding notes just in case because I don't have time to work through every scenario)
- I'm getting crashes on some features, possibly because the battery is low. I likely won't come back and update this unless someone replies (it's probably a different issue entirely, but also likely due to the battery being low), but I highly doubt it's related. If this happens and I haven't updated this message, try charging the battery if you start getting this issue especially if it's your first use.

Volleyoghurt, 2021-05-01:
> Devices detected: 0 DFU, 0 HackRF
> ERROR: HackRF not found.
> Please connect a HackRF One to your computer's USB port.
> If using a PortaPack, please put it in HackRF mode after connecting it.
> but i am [...]

this works for me:
https://github.com/sharebrained/portapack-hackrf/wiki/Updating%20Firmware%20Windows

YourBrothe, 2022-06-13:
> Devices detected: 0 DFU, 0 HackRF ERROR: HackRF not found. Please connect a HackRF One to your computer's USB port. If using a PortaPack, please put it in HackRF mode after connecting it. but i am [...]
 Can't open device Bus 002 Device 004: ID 0a5c:5800: Entity not found ans

YourBrothe, 2022-06-13:
Can't open device Bus 002 Device 004: ID 0a5c:5800: Entity not found

YourBrothe, 2022-06-13:
Can't open device Bus 002 Device 004: ID 0a5c:5800: Entity not found ans

YourBrothe, 2022-06-13:
Can't open device Bus 002 Device 004: ID 0a5c:5800: Entity not found  
give me answer