sharebrained/portapack-hackrf #160: Cannot install portapack-hackrf firmware

philsalesses, 2020-06-06:
I am on Mac OSX Catalina 10.15.4.

I plugged the HackRF into the USB port and verified the operating system can find it. I followed instructions on HackRF repository to install the newest firmware.

```MyComputer:build $ hackrf_info
hackrf_info version: git-43e6f99*
libhackrf version: git-43e6f99* (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000blahblah
Board ID Number: 2 (HackRF One)
Firmware Version: git-43e6f99 (API:1.04)
Part ID Number: 0xa000cb3c 0x00584763
```

Then I downloaded the firmware from this repo.

```MyComputer:portapack-hackrf $ cd ~/Desktop/portapack-h1-firmware-20190405/
MyComputer:portapack-h1-firmware-20190405 $ dfu-util --device 1fc9:000c --download hackrf_usb.dfu --reset
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Deducing device DFU version from functional descriptor length
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[=========================] 100%        35188 bytes
Download done.
dfu-util: unable to read DFU status after completion
dfu-util: can't detach
Resetting USB to switch back to runtime mode
```
This last command never competes and hangs there. The lights on the HackRF go back to what looks like runtime mode. However my computer cannot find the HackRF connected to my USB.

If I open a new terminal window while the firmware is running, and run the next command:

```MyComputer:portapack-h1-firmware-20190405 $ hackrf_spiflash -w portapack-h1-firmware.bin
File size 325092 bytes.
hackrf_open() failed: HackRF not found (-5)
MyComputer:portapack-h1-firmware-20190405 $ 
```

Now I'm stuck. Please help.

philsalesses, 2020-06-06:
If I try to build the firmware from scratch, it bonks too.

Inside /firmware/CMakeList.txt I changed line 55 from 
`COMMAND sleep 3s`
to 
`COMMAND sleep 15s`

Then try to make firmware nad it bonks because I'm not on Debian. Macs don't have the xsvf command...

```
MyComputer:build $ make firmware
[  1%] Generating hackrf_cpld_data.cpp
/bin/sh: --xsvf: command not found
make[3]: *** [application/hackrf_cpld_data.cpp] Error 127
make[2]: *** [application/CMakeFiles/application.elf.dir/all] Error 2
make[1]: *** [CMakeFiles/firmware.dir/rule] Error 2
make: *** [firmware] Error 2
```

philsalesses, 2020-06-06:
Usually, when I plug in the portapack/HackRF into USB, it flashes white very quickly and then goes back to black. However, one or two cables I found were able to produce this screen. I have no idea how to reproduce it. Sometimes when I plug in the cable, it gives me this. It's still unrecognizable to the OS, but it's something different.

[image]

philsalesses, 2020-06-06:
I got it to work. Once it showed this screen, I made sure it was no longer in DFU mode and I ran 

```
MyComputer:portapack-h1-firmware-20190405 $ hackrf_spiflash -w portapack-h1-firmware.bin
File size 325092 bytes.
Erasing SPI flash.
Writing 325092 bytes at 0x000000.
```
I unplugged it and then it worked. Hope this helps someone else.

closed 2020-06-06 by philsalesses

eried, 2020-06-06:
What you did different to make it work? I am trying to document this type of [procedures](https://github.com/eried/portapack-mayhem/wiki/Update-firmware#alternative-environment) for beginners (my main question is if step 5 and 6 in my link would solve the problem you had - I know it is not for mac, but I guess same applies)

sck-nogas, 2020-06-07:
> I got it to work. Once it showed this screen, I made sure it was no longer in DFU mode and I ran
>
> ```
> MyComputer:portapack-h1-firmware-20190405 $ hackrf_spiflash -w portapack-h1-firmware.bin
> File [...]

Well, this is my install "script" for Mayhem my Mac 10.15.5

alias portapak-mayhem='cd ~/src/portapack-mayhem/ && git pull && rm -rf build && git submodule update --init --recursive && export PATH=~/bin/gcc-arm-none-eabi-9-2019-q4-major/bin:${PATH} && mkdir build && cd build && cmake .. && make firmware && hackrf_spiflash -w firmware/portapack-h1_h2-mayhem.bin'

This will do all the things to keep me on the most recent git version. It's what I've been doing for several years with portapak/Havoc/Mayhem

The only thing I have to do every 9-12 months is update the ARM compilers and change the PATH entry to match the new install

bitfarmer, 2020-07-01:
Yesterday I flashed my HackRF+Portapack using my Mac and it wasn't too difficult although the normal instructions do lead to unexpected hangs.

While I do run Debian/Devuan and even Windows sometimes, I like things to just work on my Mac. I am running macOS Catalina 10.15 but the actual OS version is irrelevant here.

I placed the HackRF in DFU mode and checked it was visible to the computer:

```
dfu-util -l
```
```
Found DFU: [1fc9:000c] ver=0100, devnum=6, cfg=1, intf=0, path="20-3", alt=0, name="DFU", serial="ABCD"
```

*Somewhere in the depths of my mind I recalled this problem was previously solved by adding `--alt 0` to the dfu command. This problem did happen when I originally flashed the HackRF, even before I had a Portapack attached to it*.

Flashing it:

```
dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb.dfu
```

and afterwards, without any intervening reset, issuing:

```
hackrf_spiflash -w portapack-h1-firmware.bin
```

worked without any problems.

I hope it will help someone who is in a bind.

EI7DKB

Stans11, 2023-06-21:
> I am on Mac OSX Catalina 10.15.4.
> 
> I plugged the HackRF into the USB port and verified the operating system can find it. I followed instructions on HackRF repository to install the newest firmware.
> 
> ```
> MyComputer:build $ hackrf_info
> hackrf_info version: git-43e6f99*
> libhackrf version: git-43e6f99* (0.5)
> Found HackRF
> Index: 0
> Serial number: 0000000000000000blahblah
> Board ID Number: 2 (HackRF One)
> Firmware Version: git-43e6f99 (API:1.04)
> Part ID Number: 0xa000cb3c 0x00584763
> ```
> 
> Then I attached the Portapack to the HackRF One according to instructions here: https://github.com/sharebrained/portapack-hackrf/wiki/Assembly
> 
> and again ran `$ hackrf_info` to see if it's still recognized.
> 
> It was not.
> 
> ```
> MyComputer:build $ hackrf_info
> hackrf_info version: git-43e6f99*
> libhackrf version: git-43e6f99* (0.5)
> No HackRF boards found.
> ```
> 
> I continued anyway by downloading the firmware from this repo: https://github.com/sharebrained/portapack-hackrf/wiki/Updating-Firmware
> 
> and followed the instructions to install the firmware on to the H1.
> 
> ```
> MyComputer:portapack-hackrf $ cd ~/Desktop/portapack-h1-firmware-20190405/
> MyComputer:portapack-h1-firmware-20190405 $ dfu-util --device 1fc9:000c --download hackrf_usb.dfu --reset
> dfu-util 0.9
> 
> Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
> Copyright 2010-2016 Tormod Volden and Stefan Schmidt
> This program is Free Software and has ABSOLUTELY NO WARRANTY
> Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
> 
> dfu-util: DFU suffix CRC does not match
> dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
> Deducing device DFU version from functional descriptor length
> Opening DFU capable USB device...
> ID 1fc9:000c
> Run-time device DFU version 0100
> Claiming USB DFU Interface...
> Setting Alternate Setting #0 ...
> Determining device status: state = dfuIDLE, status = 0
> dfuIDLE, continuing
> DFU mode device DFU version 0100
> Device returned transfer size 2048
> Copying data from PC to DFU device
> Download	[=========================] 100%        35188 bytes
> Download done.
> dfu-util: unable to read DFU status after completion
> dfu-util: can't detach
> Resetting USB to switch back to runtime mode
> ```
> 
> This last command never exits and hangs there. The lights on the HackRF go back to what looks like runtime mode. However, my computer still cannot find the HackRF connected to my USB in either runtime or DFU modes.
> 
> If I open a new terminal window while the firmware is running, and run the next command:
> 
> ```
> MyComputer:portapack-h1-firmware-20190405 $ hackrf_spiflash -w portapack-h1-firmware.bin
> File size 325092 bytes.
> hackrf_open() failed: HackRF not found (-5)
> MyComputer:portapack-h1-firmware-20190405 $ 
> ```
> 
> Now I'm stuck. Please help.

IM USING WINDOWS BUT GOT LINUX UBUNTU SOFTWARE BUT CANT CONNECT EITHER