sharebrained/portapack-hackrf #162: Building firmware on Ubuntu fails because of the assert macro.

Nurrl, 2020-06-27:
*The issue is in the title ^*

## My evironment
#### Ubuntu version
```shell
$ lsb_release -dc
Description:	Ubuntu 20.04 LTS
Codename:	focal
```
#### Building tools
```shell
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1)

$ cmake --version
cmake version 3.16.3

$ dfu-util --version
dfu-util 0.9
```

## The actual problem
```shell
$ git clone --recurse-submodules https://github.com/sharebrained/portapack-hackrf.git
Cloning into 'portapack-hackrf'...
*Trucated*

$ cd portapack-hackrf && mkdir build && cd build && cmake ..
*Trucated*
-- Build files have been written to: /tmp/portapack-hackrf/build

$ make firmware
*Truncated*
[ 15%] Building CXX object firmware/baseband/CMakeFiles/baseband_shared.dir/baseband.cpp.obj
In file included from /usr/include/newlib/sys/reent.h:503,
                 from /usr/include/newlib/stdlib.h:18,
                 from /usr/include/newlib/c++/9.2.1/cstdlib:75,
                 from /usr/include/newlib/c++/9.2.1/bits/stl_algo.h:59,
                 from /usr/include/newlib/c++/9.2.1/algorithm:62,
                 from /tmp/portapack-hackrf/firmware/common/utility.hpp:28,
                 from /tmp/portapack-hackrf/firmware/common/lpc43xx_cpp.hpp:29,
                 from /tmp/portapack-hackrf/firmware/baseband/baseband.cpp:24:
/tmp/portapack-hackrf/firmware/common/lpc43xx_cpp.hpp:72:13: error: expected unqualified-id before ')' token
   72 | inline void assert() {
      |             ^~~~~~
/tmp/portapack-hackrf/firmware/common/lpc43xx_cpp.hpp:72:13: error: expected ')' before '?' token
   72 | inline void assert() {
      |             ^~~~~~
make[3]: *** [firmware/baseband/CMakeFiles/baseband_shared.dir/build.make:908: firmware/baseband/CMakeFiles/baseband_shared.dir/baseband.cpp.obj] Error 1
make[2]: *** [CMakeFiles/Makefile2:972: firmware/baseband/CMakeFiles/baseband_shared.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:647: firmware/CMakeFiles/firmware.dir/rule] Error 2
make: *** [Makefile:326: firmware] Error 2
```

Here you can see the preprocessor tries to replace your function definition by the `assert` macro, which breaks the compilation

## The fix

```diff
--- a/firmware/common/utility.hpp
+++ b/firmware/common/utility.hpp
@@ -26,6 +26,7 @@
 #include <cstdint>
 #include <cstddef>
 #include <algorithm>
+#undef assert
 #include <complex>
 #include <memory>
 ```

With the `#undef` the preprocessor doesn't mess up the code and it compiles well :)

Nurrl, 2020-06-27:
I can make a pull request if you want.

YangWang92, 2020-07-12:
I got the same problem on Ubuntu, and thanks for @Nurrl's solution!