sharebrained/portapack-hackrf #167: Hardware memory button cell redundant

PapaPendragon, 2020-08-18:
When storing settings on the portapack you have to use a 1,5V back-up button-cell placed in it's holder. Then you have to sandwitch the portapack with the hackrf and put them in a casing screwing all to gether. A solution of this is to use the battery pack (3,7V) for power suppy of the settings you've made. The 3,7V battery whitch provides the whole assy with power and is rechargeable, will then also deliver the power for the 1.5V button-cell as replacement. How? solder a 22K resistor on the positive pole of the buttoncell holder and wire from the other side of the resistor to the positive pole of the connector of the 3,7V rechargeable battery. The current flow I estimated is about 1mA, so providing 1,5V for the need of remembering your settings like normally the buttoncell did this work for you. But then a time will come to replace that dammed buttoncell again. No way.  Greetz PA3CMP

jboone, 2020-08-19:
Thanks. The PortaPack H1 design in this source repository does not have a rechargeable battery. If there is another repository containing a PortaPack hardware design with a rechargeable battery, I'm sure they would appreciate your input.

closed 2020-08-19 by jboone

PapaPendragon, 2020-08-19:
Tnx for reply, 

Ok! but the hardware Portapack board (and most of them) have a coin-cell holder, and that's a hardware design item.
Neverless I don't use that and have now perfected it (the resistor of 2K2 is changed for one of 470 𝛀) and now most of all settings I've made on softwarebase are remembered. Anyway date/time and interface (UI-settings), and some settings in receive and transmit mode will remembered. But not all of them! So there lies a software design challange.
In order your suggestion, I will place an shift this repository to another fora.

Greetz Ralph