sharebrained/portapack-hackrf #175: i know it's been brought up before, but i can't find an answer that fixes my issue...

superchuckles, 2021-03-01:
ok, here's what i've got - hackrf, works great, correct drivers installed, etc.  bought a portapack (ver2), installed mayhem firmware.  portapack works great, except, when i put it in hackrf mode -- computer can't see it.  if i take it apart & unplug the portapack, hackrf works fine, computer  can see it, can tune in various things, etc.  plug the portapack back in, portapack works, but when i put in hackrf mode its screen turns blue with bunch of symbols on the left side & one that looks like a sine wave on the bottom right with a circle around it, but computer can't see it.  it's getting tiresome having to disassemble & reassemble it over & over.  is there some pin i can cut & manually install a switch on or something?  btw, i've tried it with original portapack firmware too, same thing

eried, 2021-03-02:
I think is a problem in the usb cable

superchuckles, 2021-03-03:
but you'd be wrong.  usb cable works just fine as soon as i unplug portapack

eried, 2021-03-03:
I remembered someone asked the same somewhere and he solved it too! haha but I cant recall. I would suggest asking in our [discord](https://discord.gg/tuwVMv3) if nobody else has a clue.

kooskwak, 2021-04-27:
Had the same problem last week. My problem was indeed the USB cable!