sharebrained/portapack-hackrf #184: black screen on portapack H2 after firmware update

adammanella, 2022-05-03:
updated to 1.5.1 and now I have a black screen on my H2.
tried the nightly build as well with the same issue.
anyone having similar issues?
I am running OSX
Thank you,
A

adammanella, 2022-05-03:
the only issue I was getting came after flashing in dfu mode...... 

dfu-util: unable to read DFU status after completion (LIBUSB_ERROR_IO)

adammanella, 2022-05-03:
SOLVED!

If your device won't boot and leaves you on a black screen, then you will need to power off the device and then hold down the LEFT button while you power it on. This could take up to 10 seconds.

https://github.com/eried/portapack-mayhem/wiki/Won't-boot

mikachuman, 2022-11-28:
> SOLVED!
> 
> If your device won't boot and leaves you on a black screen, then you will need to power off the device and then hold down the LEFT button while you power it on. This could take up to 10 seconds.
> 
> https://github.com/eried/portapack-mayhem/wiki/Won't-boot

Tried this and now instead of a black screen, I'm seeing white only. H1